earlephilhower/pico-quick-toolchain #2: CMSIS-DAP adapter support and more current openocd source code?

majbthrd, 2021-07-01:
Would you please consider adding CMSIS-DAP adapter support (add --enable-cmsis-dap in addition to --enable-picoprobe to configure) to the next build of OpenOCD?

Also, there are more current source code and configuration files available with added capabilities available upstream from the openocd fork currently used for the build.

In particular, board/pico-debug.cfg and target/rp2040-core0.cfg are the two script configuration files that would be desirable to have.  These and CMSIS-DAP support would allow arduino-pico to do sketch uploads in the same manner that it presently does with picoprobe, but without picoprobe’s dependency on two RP2040 boards.

earlephilhower, 2021-07-01:
Sure thing.  So, what's in https://github.com/raspberrypi/openocd/commit/71510a77a61c6eb2b1266e00010f8f258785a54b should be good?

majbthrd, 2021-07-01:
Much appreciated!  Yes, that commit should be good.  It shows as the build failing, but I believe that is a build system issue that was only corrected on a different branch.  If I am reading the tea leaves correctly, RPi is punting on changing this until they've merged RPi's fork with the upstream genuine OpenOCD (which has evolved considerably in the interim).

Also, I should have originally noted that one gotcha to adding CMSIS-DAP could be the added libraries.  Depending on the Linux target, HIDAPI may or may not be required.  macOS requirements seem to vary (and I noticed that you are building macOS openocd separately and including it in blobs).

earlephilhower, 2021-07-01:
Good to know.  I have to manually build the Win and Mac OpenOCDs because getting the cross-compiler to have the necessary libusb/etc. cross-dependencies is something I was just not able to get to work.  The rest of the toolchain, no problem.  One `make` on my server and 30 mins later I have 6 different archs built and ready.

majbthrd, 2021-07-02:
I keep remembering more things that I should have included initially... here are two other pieces of information that might be of help:

I have some Linux build instructions here (note the libhidapi-dev dependency):

https://github.com/majbthrd/pico-debug/blob/master/howto/openocd.md

and Mac users have posted their own tips here:

https://github.com/majbthrd/pico-debug/issues/5

earlephilhower, 2021-07-02:
Argh, it's not as clean as that.  The `rp2040` branch where you have your updates does not contain the `picoprobe` debugger interface.

I'll try merging the `picoprobe` and `rp2040` branches and see what comes out.

I don't understand why the RPI org wouldn't do the same and then push everything upstream.  It *looks* like the picoprobe stuff is a pretty standalone addition.

majbthrd, 2021-07-02:
Argh.  Heaven only knows why RPi thought it necessary to introduce picoprobe in the first place.  There are established open-source adapters standards like CMSIS-DAP, but no, RPi had to create something proprietary.

closed 2021-07-05 by earlephilhower

earlephilhower, 2021-07-05:
It's surprisingly hard to include add'l dependencies for cross-compiling.  HIDAPI ended up needing UDEV which can't be cross-installed so you have to jump through hoops to get it into the x-compile link stages.

I'll have a new toolset up later today.  Could you give it a quick check when available?

majbthrd, 2021-07-05:
Terrific!  I've done the build locally and grafted in the newer openocd files into pico-arduino for testing.  All signs point to CMSIS-DAP working as expected.

commit 7855b0c9 (2021-07-05): Merge pull request #3 from earlephilhower/cmsis

Add CMSIS-DAP build for OpenOCD

Fixes #2