earlephilhower/pico-quick-toolchain #4: Issue when using Picoprobe 'libusb_bulk_write error'

maxgerhardt, 2021-07-16:
When I use the latest `x86_64-w64-mingw32.openocd-e3428fadb.210706` from this repo and fire up OpenOCD to connect via the Picoprobe, I at first get a successfull connection, but then every 3 seconds there's an error message which finally results in OpenOCD shutting down and the previously lit-up light on the picoprobe to turn off.

```
C:\Users\Max\Downloads\x86_64-w64-mingw32.openocd-e3428fadb.210706\openocd>bin\openocd.exe -f interface/picoprobe.cfg -f target/rp2040.cfg
Open On-Chip Debugger 0.10.0+dev-ge3428fadb-dirty (2021-07-06-00:38)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 5000 kHz
Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: libusb_bulk_read error: LIBUSB_ERROR_TIMEOUT
Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: libusb_bulk_read error: LIBUSB_ERROR_TIMEOUT
Error: libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: libusb_bulk_read error: LIBUSB_ERROR_TIMEOUT
Info : DAP init failed

Assertion failed!

Program: C:\Users\Max\Downloads\x86_64-w64-mingw32.openocd-e3428fadb.210706\openocd\bin\openocd.exe
File: src/jtag/drivers/picoprobe.c, Line 100

Expression: pkt_hdr->total_packet_length <= PICOPROBE_MAX_PACKET_LENGTH
``` 

Verbose output (`-d3`) 

```
...
Debug: 338 825 arm_dap.c:116 dap_init_all(): Initializing all DAPs ...
Debug: 339 825 picoprobe.c:452 picoprobe_swd_switch_seq(): DORMANT-to-SWD
Error: 340 10830 libusb_helper.c:250 jtag_libusb_bulk_write(): libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Error: 341 20840 libusb_helper.c:267 jtag_libusb_bulk_read(): libusb_bulk_read error: LIBUSB_ERROR_TIMEOUT
Debug: 342 20840 adi_v5_swd.c:656 swd_multidrop_select_target(): Failed to select core 0, retrying...
Error: 343 30860 libusb_helper.c:250 jtag_libusb_bulk_write(): libusb_bulk_write error: LIBUSB_ERROR_TIMEOUT
Debug: 345 40870 adi_v5_swd.c:656 swd_multidrop_select_target(): Failed to select core 0, retrying...
...
```

When I use the latest version of https://github.com/raspberrypi/openocd/ in the picoprobe branch, and I use the older libusb dll version as per https://github.com/raspberrypi/picoprobe/issues/3, I don't get this error. 

```
..
Debug: 336 1058 command.c:146 script_debug(): command - dap init
Debug: 338 1060 arm_dap.c:116 dap_init_all(): Initializing all DAPs ...
Debug: 339 1063 picoprobe.c:452 picoprobe_swd_switch_seq(): DORMANT-to-SWD
Info : 340 1068 adi_v5_swd.c:226 swd_multidrop_connect(): SWD DPIDR 0x0bc12477
Info : 341 1071 adi_v5_swd.c:227 swd_multidrop_connect(): SWD DLPIDR 0x00000001
Debug: 342 1074 arm_adi_v5.c:653 dap_dp_init(): rp2040.core0.dap
Debug: 343 1077 arm_adi_v5.c:698 dap_dp_init(): DAP: wait CDBGPWRUPACK
Debug: 344 1079 arm_adi_v5.h:511 dap_dp_poll_register(): DAP: poll 4, mask 0x20000000, value 0x20000000
Debug: 345 1084 arm_adi_v5.c:706 dap_dp_init(): DAP: wait CSYSPWRUPACK
Debug: 346 1087 arm_adi_v5.h:511 dap_dp_poll_register(): DAP: poll 4, mask 0x80000000, value 0x80000000
Debug: 347 1091 picoprobe.c:452 picoprobe_swd_switch_seq(): DORMANT-to-SWD
..
```

Note though that once I use the OpenOCD from this version with the above command; I do need to re-plug the device for it to be usable again, otherwise I *do* run in the timeout errors.

Maybe something went wrong in the merge as detailed in #2 or the two branches aren't in fact mergable without these side-effects. 

Side Info: OS is Windows 10 x64, the Picoprobe firmware comes directly from [raspberrypi.org](https://www.raspberrypi.org/documentation/rp2040/getting-started/static/fec949af3d02572823529a1b8c1140a7/picoprobe.uf2), I used Zadig to install libusb-win32 drivers for the device.

maxgerhardt, 2021-07-18:
I actually just tried all OpenOCD versions released here, 
* `x86_64-w64-mingw32.openocd-e3428fadb.210706`
* `x86_64-w64-mingw32.openocd-d58c2ef5e.210629`
* `x86_64-w64-mingw32.openocd-d58c2ef5e.210607`
* `x86_64-w64-mingw32.openocd-d58c2ef5e.210328`

And none could connect to the Pi Pico with the picoprobe adapter, all with the same error message as per above.

I'm attching a zip compressed version of my Windows build here for comparison, which in the same setup is able to connect to the Pico perfectly fine.

[openocd_raspbi_picoprobe.zip](https://github.com/earlephilhower/pico-quick-toolchain/files/6837054/openocd_raspbi_picoprobe.zip)

earlephilhower, 2021-07-19:
That is very strange.  While I did not test the absolute latest release which was cross-compiled from Linux, the prior one was built in a Windows VM manually (win32 only, both x64 and x32 versions are the same exe/dlls) and I an almost 100% certain that basic debugging worked for me.

This'll be in the queue.  I honestly don't mind just repackaging your version as a blob if I can't get the cross-compile/VM-built one running, but I would like to verify it's failing everywhere and at least try and figure out WTH is going on...

maxgerhardt, 2021-07-19:
Definitely strange. 

Just for confirmation, I've tried selectin the Pi Pico (Picoprobe) board with your latest released Arduino core, no modifications, and uploading gives the same result.

[image]

Replacing the OpenOCD with my precompiled version gives

[image]

..a successfull connection to the board, but OpenOCD can't find the referenced file due to no / wrong escaping in the `-c program {file}` part of the command :/ (I'll open a separate issue or PR for this is in the core).

Also be aware when testing in a Windows VM that I've used Zadig to load **WinUSB** drivers.  In the issue linked 2 posts above there's a discussion that loading libusb-win32 drivers leads to a crash (segfault) when using `libusb-1.0.dll` of the latest version (1.0.24), whereas loading WinUSB drivers does not.

And my final thought is that the OpenOCD version might be linked with a particular picoprobe firmware. I'm using the latest picoprobe.uf2 from https://www.raspberrypi.org/documentation/rp2040/getting-started/#board-specifications with the wireup as stated in https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf Appendix A.

maxgerhardt, 2021-09-21:
@earlephilhower have you had time to look into this (OpenOCD version released here does not work on Windows)? If we get that resolved we can make the final push for back-integration into the official PlatformIO repos (`platform-raspberrypi`)

closed 2022-02-17 by earlephilhower