earlephilhower/pico-quick-toolchain #6: Support on armv7 platform like Pi 3B+

LockonS, 2021-08-13:
I am working on a project which requires rapidly adjustment of the program, so I put an Raspberry pi 3B+ on the device as a remote building and debug platform. The building process is all good on both MacOS and Ubuntu running on an Pi4, I got an error on Pi 3B+ while building with platformio.

`Could not find the package with 'maxgerhardt/toolchain-pico @ ~5.100300.0' requirements for your system 'linux_armv7l'`

I have checked the release page and there's no pre-build package for armv7 platform yet. Will armv7 platform be supported or I need to build the package from the source code?

maxgerhardt, 2021-11-09:
This repo explicitly has https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.1-a/arm-linux-gnueabihf.arm-none-eabi-7855b0c.210706.tar.gz, does that not work?

earlephilhower, 2022-02-17:
Raspberry Pi builds are under the link Max posted, closing.

closed 2022-02-17 by earlephilhower