earlephilhower/pico-quick-toolchain #8: ARM64 macOS builds

tjhorner, 2022-04-07:
At the moment it looks like there are no ARM64 (Apple Silicon) builds available from the release page. It would be nice to have these available so I don't need to build them myself.

earlephilhower, 2022-04-07:
There's an aarch64 build for 64-bit ARM.  I don't know if MacOS supports a Linux compat layer, though, to use it.  I'm also pretty sure the Arduino IDE doesn't know what to do w/64b Mac on ARM, either (i.e. even for Win64 is still installs the 32b binaries!).

I do everything via a cross build process and haven't had a Mac since 1996, so if this is important I'll need some assistabnce.

If you look at https://github.com/earlephilhower/gcc-cross and can add in something that will let Linux x64 build Mac64ARM bits, I'd be happy to include it in the standard release process.

tjhorner, 2022-04-07:
> I don't know if MacOS supports a Linux compat layer, though, to use it.

macOS does not support ELF binaries, unfortunately, so those builds won't work.

> If you look at https://github.com/earlephilhower/gcc-cross and can add in something that will let Linux x64 build Mac64ARM bits, I'd be happy to include it in the standard release process.

Sure, I can take a look.

FWIW, I am using PlatformIO and followed the [instructions](https://arduino-pico.readthedocs.io/en/latest/platformio.html#current-state-of-development), but it just hung there for a while not doing anything, so I thought something went wrong due to platform architecture. It turns out I was just impatient, and everything took a while to clone and install.

The existing x86_64 builds do work on ARM64 (probably using Rosetta to translate it), so in the meantime that will work. But would be nice to have native builds Eventually™️

louisadamian, 2022-04-28:
Hello am looking to build for the pico on Mac. There is support for arm-eabi-gcc on macOS with a [package from Homebrew ](https://formulae.brew.sh/cask/gcc-arm-embedded). but it only supports gcc 11. Would GCC 11.3 support be possible? Thanks in advance.

earlephilhower, 2022-04-28:
It probably would work fine with pico-arduino but isn't something I can guarantee or test.  You'd need to make adjustments to the platform.txt to use the system `arm-eabi-gcc` instead of the packaged one.

earlephilhower, 2022-06-13:
Thanks, @LeSuedois.  Unfortunately, I don't think it's an M1 owner I need assistance from here.  The issue is that I cross-compile everything, from Raspberry Pi to Win32 to Mac x64 to AARCH64, all in a Docker container on a Linux system.  So what needs to be done is figure out how to cross-compile for the M1/MacOS, not actually run builds on one.

It's been a while since I built the container, but IIRC it depends on a manually cross-built XCode.  If Apple is still doing XCode, then it's probably a matter of doing the same to the latest bundle.

zjwhitehead, 2022-06-17:
In the meantime we should try to get Mac x64 version running on M1. Right now it just blows up with 
`Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ ~5.100300.0' requirements for your system 'darwin_arm64'`
Rosetta emulation works pretty well for most apps. Is there a flag or a way to force PlatformIO to just use the X64 version of the toolchain package?

earlephilhower, 2022-06-17:
>  Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ ~5.100300.0' requirements for your system 'darwin_arm64'

Yup, that'd be the problem right there...it is looking for `darwin_**arm64**` and not falling back to `darwin`.  What app are you running?

zjwhitehead, 2022-06-18:
> What app are you running?
I'm running PlatformIO with the latest config that looks like 
```
[env:OpenPPG-CRP2040-SP140]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = sparkfun_promicrorp2040
framework = arduino
board_build.core = earlephilhower

board_build.filesystem_size = 14M ; 14 Mbyte for filesystem and 2 Mbyte for program
```
I see on the PIO registry that as expected its not showing a version for Mac Arm64
https://registry.platformio.org/tools/earlephilhower/toolchain-rp2040-earlephilhower/compatibility

earlephilhower, 2022-06-18:
Ah, I see.  Arduino doesn't know what `darwin_amd64` is...it doesn't even know what `winx64` is and that's waaaay older. :(

Is there an override to specify the PIO architecture?  the Mac x64 binaries are good and should run (as they do in the IDE) but just a little slower since evidently Rosetta 2 is pretty good stuff.

Alternatively, I might be able to publisy the Mac x64 binaries as Mac ARM64 but that seems like cheating and might cause some grief upstream...

earlephilhower, 2022-06-18:
Looks like work is ongoing for M1 XCode support:  https://github.com/tpoechtrager/osxcross

earlephilhower, 2022-11-29:
I did some exploration and while I can build most tools for the M1 using the latest `osxcross`, GCC itself isn't yet compatible as a host on the Apple M1 as of now (we need an Apple ARM M1 host, a ARM M0+ target build to make RP2040 binaries).   See https://gcc.gnu.org/bugzilla/show_bug.cgi?id=96168

It looks like only 12.X will support it, and as far as I can see on the GCC changelogs there is no Apple M1 support in as of the 12.2 release.  If we move forward to 12.X and the M1 support is merged, I'll revisit.  For now, though, unfortunately it's a "hurry up and wait" operation.

zjwhitehead, 2022-11-29:
FWIW heres what my M1 Pro shows currently
```
➜  ~ gcc --version
Apple clang version 13.1.6 (clang-1316.0.21.2.5)
Target: arm64-apple-darwin21.4.0
```

earlephilhower, 2022-11-29:
Yup, that's `clang` disingenuously pretending to be `gcc`.  I'm sure GCC will be brought to the M1 eventually, it'll just take some time...

louisadamian, 2022-12-12:
The patching is done with in this Ruby file https://github.com/Homebrew/homebrew-core/blob/master/Formula/gcc.rb but when I try to build this from source I get an error. There are also binaries available which is what is installed by default and what I am using now and they are available here https://github.com/homebrew/homebrew-core/pkgs/container/core%2Fgcc%2F12

earlephilhower, 2022-12-14:
Thanks!  I looked at it and it seems like no patching is needed on the 12.2 release of GCC.  `darwin20` or `darwin21` looks like it's been added.

Unfortunately that's only 1/2 of what I need.  Because I build on Linux I actually need *2* cross-compilers built.
1. Linux x86_64 **host**, AARCH64-darwin20 **target**
2. Using (1), build a AARCH64-darwin20 **host**, Cortex-M0+ **target** compiler.

(1)  comes from OSXCross which presently only builds `clang` for the Arm, not `gcc`.  I'm not sure that `clang` can be used as the compiler for GCC (it's not used in any of the other cross-builds I do).  Will need to see if that's supported.

earlephilhower, 2022-12-23:
It turns out there **are** many patches required to make GCC be able to target `aarch-apple-darwin2x`.  The Ruby file downloads a clean tarball and then applies patches a **20K++ line patch** from https://github.com/Homebrew/formula-patches/tree/master/gcc

Only GCC 11.3, 12.1, and 12.2 are patched, so the GCC 10 we're using is not supported at all.

I attempted to apply the patch and run an `osxcross` build for GCC 12.2, and it throws massive amounts of compile warnings and fails while build a GCC prerequisite.  (On the existing cross builds I don't believe there are more than a handful of warnings thrown for the entire process).

So, that means that even (1) above isn't available,yet.

I'll leave this open but I think this is not going to happen until GH enables their own M1 hosted runners so I can special-case the M1 build...

louisadamian, 2022-12-23:
It looks like there is a arm64 host version of the arm baremetal toolchain available now https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads so that might be a better option.

earlephilhower, 2022-12-23:
No, I already build `aarch64-linux` just fine.  It's the Apple specific one (`aarch64-apple-darwinXX`) which is the problem, sadly.

Jason2866, 2023-01-03:
Maybe this https://github.com/espressif/gcc/commits/esp-2021r2-patch4
does help, espressif used my cumulative patch to build the native esp32x M1 toolchains for gcc 8.4. Built with https://github.com/espressif/crosstool-NG/tree/esp-2021r2-patch5

nicco, 2023-03-20:
Hey, having the same issue, any update on this?

nicco, 2023-03-20:
> macOS does not support ELF binaries, unfortunately, so those builds won't work.
>
>
> Sure, I can take a look.
>
> FWIW, I am using PlatformIO and followed the [...]

I'm getting this error:
Reading symbols from /Users/niccoloperra/Projects/tesla-hud/.pio/build/pico/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = cmsis-dap
PlatformIO: Initializing remote target...
dyld[11467]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <B9B4AFBD-16C1-38F7-8A32-08215C78F03B> /Users/niccoloperra/.platformio/packages/tool-openocd-rp2040-earlephilhower/bin/openocd
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
  
 Did you have any similar issues?

earlephilhower, 2023-03-20:
You need the Intel format USB library and only have the ARM64 version.

Back in the day, when going from 68k->PPC and IIRC PPC->x86, the Mac had "fat binaries" with code for both architectures in it, so a library or app could run on either architecture.  Does it have it for the x86_64->arm64 transition?  Can you install x86_64 versions of libraries in some other directorylike /usr/local/opt/x86_64 or something (like Linux uses for multiarch)?

earlephilhower, 2024-01-27:
Fixed by #45

closed 2024-01-27 by earlephilhower