earlephilhower/pico-quick-toolchain #16: openocd won't run on vanilla OSX (ARM) anymore

fietser28, 2022-09-28:
Using the latest openocd.osx.tar file gives the following error on OS X 12.5.1 (arm processor, using X86 binaries):

`% ./openocd/bin/openocd                
dyld[85118]: Library not loaded: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib'
  Referenced from: '/Users/jan/tmp/openocd/openocd/bin/openocd'
  Reason: tried: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/lib/libhidapi.0.dylib' (no such file), '/usr/lib/libhidapi.0.dylib' (no such file)`

This is on a reasonably vanilla machine ( homebrew not installed, Xcode is).

I tried downloading older versions and release [1.3.0-a](https://github.com/earlephilhower/pico-quick-toolchain/releases/tag/1.3.0-a) doesn't give this error, but starting from [1.3.1-a](https://github.com/earlephilhower/pico-quick-toolchain/releases/tag/1.3.1-a) it does. To me it seems something is changed in the build proces introducing this library dependency.

fietser28, 2022-09-29:
Issue is also discussed in: https://github.com/earlephilhower/arduino-pico/issues/703 

I will test the proposed workaround in that issue.

earlephilhower, 2022-12-23:
Closing as related/dup of #8 which is tracking any ARM Mac builds...they seem two sides of the same coin...

closed 2022-12-23 by earlephilhower