earlephilhower/pico-quick-toolchain #18: Enabling openocd swd interfaces for raspberry pi

codeTom, 2022-10-07:
The current configuration disables most openocd interfaces. Is there a specific reason for this beyond saving space/compilation time? On Raspberry Pi, having the ability to upload through SWD on GPIO is useful, can `sysfsgpio` and `bcm2835gpio` be enabled in openocd config?

https://github.com/earlephilhower/pico-quick-toolchain/blob/83a417da8d3470783009c16549d7ba52c7646a9a/Makefile#L274

https://github.com/earlephilhower/pico-quick-toolchain/blob/83a417da8d3470783009c16549d7ba52c7646a9a/Makefile#L281

earlephilhower, 2022-10-07:
It's been a while but space/time was one reason.  I think there were also some issues with dependencies for some of the options, too (everything I do is cross-built on my Linux box  so I don't have full access to RPI/Mac/Windows libraries that are non-standard).

If you have some you'd like to see re-added (just those two?) let me know and I can give it a try and see if things build properly.

codeTom, 2022-10-08:
Thanks, just those two to enable flashing via GPIO from raspberry pi. Good point about the dependencies, I'm hoping the openocd build system is clever enough to handle it (I can't test it at the moment, bizarrely due to disk space).

earlephilhower, 2022-10-09:
Here is a tarball generated with `--enable-<those 2 options>` set.  Looking at the config output, though, I don't think it actually added the BRCM or SYSFS bits due to dependencies.

[arm-linux-gnueabihf.openocd-e3428fadb.221009.tar.gz](https://github.com/earlephilhower/pico-quick-toolchain/files/9742344/arm-linux-gnueabihf.openocd-e3428fadb.221009.tar.gz)

Let me know if it did work, and if so I can do a PR

codeTom, 2022-10-10:
Thank you, it does include those bits and it successfully flashed my RP2040-based pi hat via SWD.

PR #19 (2022-10-25): Enable sysfsgpio and bcm2835gpio in openocd
These options allow flashing and debugging via SWD using Raspberry Pi's GPIO header. Closes #18
commit a713499d (2022-10-25): Enable sysfsgpio and bcm2835gpio in openocd (#19)

These options allow flashing and debugging via SWD using Raspberry Pi's GPIO header. Closes #18
closed 2022-10-25 by earlephilhower