earlephilhower/pico-quick-toolchain #20: [COMPILER BUG] - Request: Migrate from GCC 10.3.x to GCC 11.2.x

kholk, 2022-10-31:
There's a bug in GCC 10.3.x that hits when compiling for Cortex-M0/M0+ and it's triggered when building LVGL (8.3.x).

lvgl/src/core/lv_event.c
function: `static bool event_is_bubbled(lv_event_t * e)`

Failure point:
```
    switch(e->code) {
        case LV_EVENT_HIT_TEST:
        case LV_EVENT_COVER_CHECK:
        case LV_EVENT_REFR_EXT_DRAW_SIZE:

        /* Removing this, or adding one more case entry avoids the bug */
        case LV_EVENT_DRAW_MAIN_BEGIN: 

        case LV_EVENT_DRAW_MAIN:
        case LV_EVENT_DRAW_MAIN_END:
        case LV_EVENT_DRAW_POST_BEGIN:
        case LV_EVENT_DRAW_POST:
        case LV_EVENT_DRAW_POST_END:
        case LV_EVENT_DRAW_PART_BEGIN:
        case LV_EVENT_DRAW_PART_END:
        case LV_EVENT_REFRESH:
        case LV_EVENT_DELETE:
        case LV_EVENT_CHILD_CREATED:
        case LV_EVENT_CHILD_DELETED:
        case LV_EVENT_CHILD_CHANGED:
        case LV_EVENT_SIZE_CHANGED:
        case LV_EVENT_STYLE_CHANGED:
        case LV_EVENT_GET_SELF_SIZE:
            return false;
        default:
            return true;
    }
```

Error message:
```
{standard input}: Assembler messages:
{standard input}: Error: unaligned opcodes detected in executable segment
```

This happens when building with flags `-g -Os`, but doesn't happen with `-O2` or `-g0`.

Solution:
Using GCC 11.2.1 to build the project simply works with any cflags, as expected... but anyway GCC 12 also works.

More information:
After some research, this bug has also been encountered in other projects:
apache/incubator-nuttx : https://github.com/apache/incubator-nuttx/pull/5527
Zephyr RTOS: https://github.com/zephyrproject-rtos/zephyr/commit/166d013311cd8df2267dadd43ed9f754668bb577
(Zephyr, LVGL: https://github.com/lvgl/lvgl/issues/3425)

Requested action:
Please migrate away from GCC 10.3.x to either GCC 12.1 (tested in Zephyr), or GCC 11.2.1 (tested by me) as soon as possible.

Thanks,
Angelo

earlephilhower, 2022-10-31:
I remember discussing this in the `arduino-pico` core earlier.  The x16 switch caused a boom as you discuss.

Moving to 12.x is simple here buy a little complicated practically w/the core, and it would be predicated on the Pico-SDK building cleanly w/GCC12.  https://github.com/raspberrypi/pico-sdk/issues/1029 .  There's a PR open and I believe it's scheduled for the next minor release.  Until that's out, I think this will have to wait.

I checked 10.4 and there is only an x86-64 bugfix, so no joy just bumping a minor rev.

earlephilhower, 2023-08-24:
We're moving to GCC 12.3 as soon as testing completes with the new Makefile, so closing as fixed.

closed 2023-08-24 by earlephilhower