earlephilhower/pico-quick-toolchain #21: Error: Could not find one of 'platform.json' manifest files in the package

btwebz, 2022-11-24:
Hi
When trying to install on a rpi pico w I get this:

INFO Reading configuration /config/esphome/picow-bmp280.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing picow-bmp280 (board: rpipicow; framework: arduino; platform: https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.1-a/arm-linux-gnueabihf.arm-none-eabi-7855b0c.210706.tar.gz)
--------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.1-a/arm-linux-gnueabihf.arm-none-eabi-7855b0c.210706.tar.gz
INFO Installing https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.1-a/arm-linux-gnueabihf.arm-none-eabi-7855b0c.210706.tar.gz
Unpacking  [####################################]  100%
Error: Could not find one of 'platform.json' manifest files in the package

Running ESPHome on a Pi 4b armv7

Any help would be greatly appreciated!
Thanks,
btwebz

maxgerhardt, 2022-12-02:
The toolchain is not the platform. The platform is https://github.com/maxgerhardt/platform-raspberrypi.git. Do not set `platform = <link to toolchain>`.

If you want PlatformIO to download and **possibly** use a certain package, such as the compiler toolchain, you use [`platform_packages`](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages) in the `platformio.ini` file:

```
platform_packages =
   toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.1-a/arm-linux-gnueabihf.arm-none-eabi-7855b0c.210706.tar.gz
```

However, this download's `package.json` still declares 
```
{
   "description": "arm-none-eabi-gcc",
   "name": "toolchain-pico",
   "system": [ "linux_armv6l" ],
   "url": "https://github.com/earlephilhower/pico-quick-toolchain",
   "version": "5.100300.210706"
}
```
so it will rejcet to use it on your linux armv7l machine.

What you **could** do is locally download the file, untar it somewhere, add `"linux_armv7l"` in the `system` array, then tell PlatformIO to use the toolchain from your local folder with the `symlink://` protocol:
```ini
platform_packages =
   toolchain-pico@symlink:///home/youruser/extracted_toolchain_folder`
```

btwebz, 2022-12-02:
Thanks for your help. I think doing what you suggest is beyond my basic capabilities so I will need to wait until a v7l package becomes available. Thanks again.

earlephilhower, 2022-12-23:
Publishing this as I type, so look for it in a little bit (maybe give extra time for holidays...)

closed 2022-12-23 by earlephilhower

btwebz, 2022-12-23:
Thank you so much for publishing the new version.
Is it still located at https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.1-a/arm-linux-gnueabihf.arm-none-eabi-7855b0c.210706.tar.gz

btwebz, 2022-12-23:
I found and used this:
https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.5.0-a/arm-linux-gnueabihf.arm-none-eabi-5007782.221223.tar.gz and igot the same error message:
`INFO Reading configuration /config/esphome/picow-bmp280.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing picow-bmp280 (board: rpipicow; framework: arduino; platform: https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.5.0-a/arm-linux-gnueabihf.arm-none-eabi-5007782.221223.tar.gz)
--------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.5.0-a/arm-linux-gnueabihf.arm-none-eabi-5007782.221223.tar.gz
INFO Installing https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.5.0-a/arm-linux-gnueabihf.arm-none-eabi-5007782.221223.tar.gz
Unpacking  [####################################]  100%          
Error: Could not find one of 'platform.json' manifest files in the package`

maxgerhardt, 2022-12-23:
You can't use a toolchain package as a `platform = ...` value. If anything, you would have to go through `platform_packages = toolchain-rp2040-earlephilhower@<link>`.

**However**, the ARMv7l packages are now uploaded(https://registry.platformio.org/tools/earlephilhower/toolchain-rp2040-earlephilhower/compatibility, https://github.com/maxgerhardt/platform-raspberrypi/commit/9a13c2575e569c141bd67db2ae1ed7cb61cd9ed6. You need to either update the raspberrypi platform now or just remove it to cleanly reinstall it, then it should work with 0 modifications.
```
rm -rf ~/.platformio/platforms/raspberrypi*
```
then rebuild your project

btwebz, 2022-12-23:
Thank you so much @maxgerhardt  If it is not too much of an imposition, could I please ask you to help me get this working on ESPHome? (Which I have running alogside Home Assistant)
I would greatly appreciate any help you could give me. 

With this configuration:
`esphome:
  name: picow

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git`

I am getting: 

`INFO Reading configuration /config/esphome/picow.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing picow (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
Tool Manager: Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714
INFO Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714
Error: Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714' requirements for your system 'linux_armv7l'`

maxgerhardt, 2022-12-23:
Kill the cache.

```
rm -rf ~/.platformio/.cache
```

btwebz, 2022-12-23:
Sorry, I must be doing something wrong I just keep getting the same error message. I did kill the cache and the platform and then tried to recompile. Is there any useful information I could provide you with? Thanks again.

btwebz, 2022-12-23:
Maybe I didn't successfully kill the cache and platform? I tried using the Home Assistant terminal and then by doing a Clean build files in ESPHome

maxgerhardt, 2022-12-23:
Well you can try to remove platformio and esphome fully, then reinstall esphome which should reinstall platformio.rm -rf ~/.platformiopip3 uninstall esphome

btwebz, 2022-12-23:
Awesome! Reinstalling ESPHome worked.Thank  you so much for all your help and patience. I admire your skills and ability! Have a great day!