earlephilhower/pico-quick-toolchain #22: [Request] Compile version for 32bit - armv7l

HarvsG, 2022-12-02:
Some users will be wanting to compile pico projects from their raspberry pis, some of which are 32bit only. And even on 64bit devices 32bit Raspbian is the recommended install by the r-pi foundation - https://www.raspberrypi.com/software/operating-systems/.

As such it would be great to have a 32bit version of the toolchain.

Related issue: https://github.com/esphome/issues/issues/3854

btwebz, 2022-12-02:
I too would really appreciate  a compile for 32 bit - arm7l

earlephilhower, 2022-12-02:
Can you run `gcc -dumpmachine; gcc -v` on your system and paste the output?  It should be trivial to add, but I need to know the tuple...

HarvsG, 2022-12-02:
```
pi@greenpi:~ $ gcc -dumpmachine; gcc -v
arm-linux-gnueabihf
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 8.3.0-6+rpi1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
```

earlephilhower, 2022-12-02:
This is already there in the Makefile, and the existing binary will work (but P.Io won't install until I re-upload the entire chain.)

See https://github.com/earlephilhower/pico-quick-toolchain/pull/17

It'll be in the next toolchain build, no actual compilation changes/etc. required...

closed 2022-12-02 by earlephilhower

HarvsG, 2022-12-02:
Great that is good news. When is the next build likely to be?

maxgerhardt, 2022-12-22:
@earlephilhower  sorry to ping you directly, are the packages already uploaded and are just waiting for manual review by the PIO team, or has the push not happened

earlephilhower, 2022-12-22:
Running the Arduino toolchain on a RPi 32bit (I assume only RPi3)  must really be a test of patience, ouch.

@maxgerhardt it's not been rebuilt yet.  Since it's a big D/L for folks I wanted to also get https://github.com/earlephilhower/pico-quick-toolchain/issues/23 and maybe https://github.com/earlephilhower/pico-quick-toolchain/issues/8 in as far as changes before the update.

Worst case I'll do a build by year's end...