earlephilhower/pico-quick-toolchain #23: openocd: could it be compiled with CMSIS-DAPv2 enabled?

rgrr, 2022-12-18:
Hello,

I'm currently experimenting with CMSIS-DAPv2 for the picoprobe. During my experiments I found out, that the bundled (Windows) OpenOCD does not handle CMSIS-DAPv2.

Log should look like this:

```
Open On-Chip Debugger 0.11.0-g228ede43d-dirty (2022-11-14-09:09)
...
Debug: 25 3 configuration.c:97 find_file(): found ./cmsis-dap.cfg
Debug: 26 4 command.c:146 script_debug(): command - adapter driver cmsis-dap
Debug: 28 4 command.c:146 script_debug(): command - cmsis_dap_vid_pid 0x2e8a 0x000c 0x2e8a 0x000d 0xcafe 0x4005
...
Debug: 165 47 cmsis_dap_usb_bulk.c:156 cmsis_dap_usb_open(): found product string of 0x2e8a:0x000d 'Picoprobe CMSIS-DAP v2'
Debug: 166 47 cmsis_dap_usb_bulk.c:176 cmsis_dap_usb_open(): enumerating interfaces of 0x2e8a:0x000d
Debug: 167 53 cmsis_dap_usb_bulk.c:223 cmsis_dap_usb_open(): found interface 0 string 'Picoprobe CMSIS-DAP v2'
Info : 168 53 cmsis_dap_usb_bulk.c:324 cmsis_dap_usb_open(): Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000d, serial=E6614103E7176A23
Info : 169 54 cmsis_dap.c:837 cmsis_dap_get_caps_info(): CMSIS-DAP: SWD  Supported
Info : 170 55 cmsis_dap.c:817 cmsis_dap_get_version_info(): CMSIS-DAP: FW Version = 2.0.0
Info : 171 56 cmsis_dap.c:940 cmsis_dap_swd_open(): CMSIS-DAP: Interface Initialised (SWD)
Debug: 172 58 cmsis_dap.c:1012 cmsis_dap_init(): CMSIS-DAP: Packet Size = 512
Debug: 173 58 cmsis_dap.c:1025 cmsis_dap_init(): CMSIS-DAP: Packet Count = 8
Debug: 174 58 cmsis_dap.c:1028 cmsis_dap_init(): Allocating FIFO for 3 pending packets
Info : 175 59 cmsis_dap.c:852 cmsis_dap_get_status(): SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Error: 176 60 cmsis_dap.c:368 cmsis_dap_cmd_DAP_SWJ_Clock(): CMSIS-DAP command CMD_DAP_SWJ_CLOCK failed.
...

```

But the bundled version shows:

```
Open On-Chip Debugger 0.11.0-g610f137 (2022-02-11-13:57)
...
Debug: 25 3 configuration.c:97 find_file(): found ./cmsis-dap.cfg
Debug: 26 4 command.c:146 script_debug(): command - adapter driver cmsis-dap
Debug: 28 4 command.c:146 script_debug(): command - cmsis_dap_vid_pid 0x2e8a 0x000c 0x2e8a 0x000d 0xcafe 0x4005
...
Error: 165 309 cmsis_dap.c:260 cmsis_dap_open(): unable to find a matching CMSIS-DAP device
...

```

Any chances to include an updated version which is fixed?

I already filed this to https://github.com/maxgerhardt/platform-raspberrypi.git.  Maintainer Max Gerhardt told me, that this is the right place to go.

maxgerhardt, 2022-12-20:
Specifically, this config kills it

https://github.com/earlephilhower/pico-quick-toolchain/blob/a713499d218ee490d9cdfa004f01e2ea341ce4fa/Makefile#L254

earlephilhower, 2022-12-20:
I tried to minimize the build dependencies by stripping OpenOCD to the bare minimum.  Looks like I cut a bit too deep.  Due to dependencies on libUSB not quite working when cross-compiled under Linux this was built by hand on a Windows box.  I'll see how it comes out and if I can automate this via a GH action instead of my (fragile) manual method and post a test exe here in a few days...

maxgerhardt, 2022-12-20:
For the GD32 fork of OpenOCD in the XPack version I hacked together / massaged the Github action to run natively on the github action system: https://github.com/CommunityGD32Cores/openocd-xpack/blob/xpack-develop/.github/workflows/build.yml#L34-L81

The XPack build box system appeared to be extremely picky to me, requiring a dedicated machine, it built its own toolchain, requiring an actual Mac custom runner for doing Mac builds. I had time for none of that so I had the github actions machine execute the builds that it could make directly, see https://github.com/xpack/xpack-build-box/discussions/3#discussioncomment-3854593.

earlephilhower, 2022-12-21:
@rgrr, can you give this a try:  [openocd.zip](https://github.com/earlephilhower/pico-quick-toolchain/files/10282189/openocd.zip)

I ran it on a clean Win64 system (no msys/etc.) and it was able to connect to a PicoProbe.  I don't have anything for a CMSIS DAP so you'll need to verify yourself...

Took quite a few tried to get it working under GH actions (plus, RPi's OpenOCD fork doesn't compile cleanly under GCC12!!!) but now I have something I can plug into this toolchain repo and manually fire to get any updates...
https://github.com/earlephilhower/openocd/blob/rp2040/.github/workflows/build-win32.yml
https://github.com/earlephilhower/openocd/actions/runs/3753093145

rgrr, 2022-12-22:
Hello Earle,

I can confirm that the above version is working with CMSIS-DAPv2.  Many thanks to you for your fast reaction.

If you want to test it with picoprobe, you can use my fork at https://github.com/rgrr/picoprobe/tree/cmsis-dapv2

earlephilhower, 2022-12-22:
Thanks for the update!  I'll add in the new build process and artifacts to the next pico-quick-toolchain build coming soon.

closed 2022-12-22 by earlephilhower