earlephilhower/pico-quick-toolchain #25: Update to OpenOCD 0.12?

rgrr, 2023-02-02:
Hello Earle,

don't want to bother you (I'm the one who filed issue #23)... but would it be possible to update to openocd 0.12?

The currently used openocd 0.11 has an issue with more recent CMSIS-DAP protocol, specific > 2.0.0 or anything after CMSIS 5.7.0 which is almost 3 years old.

Problem is, that openocd 0.11 cannot handle DAP_Info/Capabilities with two bytes and thus rejects connection.
On the other hand pyocd requires a newer CMSIS-DAP version for its auto-detect feature.

```
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6614C775B1D4D35
Info : CMSIS-DAP: FW Version = 2.1.1
Info : CMSIS-DAP: Serial# = E6614C775B1D4D35
Error: CMSIS-DAP: SWD not supported
Error: No Valid JTAG Interface Configured.
```

Another point is that openocd 0.12 is now an actual release.

earlephilhower, 2023-02-02:
Sorry, I don't think that's possible.  We need to support Picoprobe which means we're stuck with the Raspberry Pi fork, rp2040 branch.  That has not been updated to 0.12.

I think your best bet is to build/get a 0.12 binary and stuff it in place of the one from this package set. Since you don't use Picoprobe it should be fine there.

You could also open a raise in the RPI OpenOCD fork and see if they'd be willing to pull upstream.

rgrr, 2023-02-02:
Ok, I didn't think about that.  I will check the possibilities ;-)
I guess we can close this then.

closed 2023-02-02 by earlephilhower