earlephilhower/pico-quick-toolchain #30: Windows MinGW binaries need non-delivered DLLs

maxgerhardt, 2023-09-17:
For the toolchain build https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.1.0-a/x86_64-w64-mingw32.arm-none-eabi-d3d2e6b.230911.zip

Per

```
$ objdump -p arm-none-eabi-gdb.exe  | grep "DLL Name:"
	DLL Name: ADVAPI32.dll
	DLL Name: KERNEL32.dll
	DLL Name: msvcrt.dll
	DLL Name: USER32.dll
	DLL Name: WS2_32.dll
	DLL Name: libgcc_s_seh-1.dll
	DLL Name: libstdc++-6.dll
```
has a dependency on at least `libstdc++-6.dll` and `libgcc_s_seh-1.dll`, the one of which in MinGW has another dependency on `libwinpthread-1.dll`. However, none of these DLLs are delivered in the downloadable ZIP file.

The toolchain build must be altered to either:
* Only build fully static binaries (`--enable-static` ?), which would blow up code size a bit
* Enable only a statically built libgcc so that the last two (three) dependencies disappear (` -static-libgcc` ?)
* Package in the missing DLLs dynamically (`cp $(which libgcc_s_seh-1.dll) $packageOutput/bin/.`  etc)

maxgerhardt, 2023-09-17:
Funnily enough (or logically?), only the GDB build seems to be affected, as the GCC binaries do **not** require these DLLs.

```
$ objdump -p arm-none-eabi-gcc.exe | grep "DLL Name:"
	DLL Name: ADVAPI32.dll
	DLL Name: KERNEL32.dll
	DLL Name: msvcrt.dll
	DLL Name: USER32.dll
```
`arm-none-eabi-gdb.exe` is the only one requiring this.

earlephilhower, 2023-09-17:
This could be a quagmire unfortunately.  Things are built already with `--enable-static`:
https://github.com/earlephilhower/pico-quick-toolchain/blob/bf1a2ae4d35a019cac7f4494c543abca900ae638/Makefile#L262 and the logs do show it being passed in
````
  $ /workdir/repo/binutils-gdb-gnu/configure --prefix=/workdir/arm-none-eabi.win64 --build=x86_64-linux-gnu --host=x86_64-w64-mingw32 --target=arm-none-eabi --disable-shared --with-newlib --enable-threads=no --disable-__cxa_atexit --disable-libgomp --disable-libmudflap --disable-nls --without-python --disable-bootstrap --enable-languages=c,c++ --disable-lto --enable-static=yes --disable-libstdcxx-verbose --disable-decimal-float --with-cpu=cortex-m0plus --with-no-thumb-interwork --with-libgmp-prefix=/workdir/arena.win64/gmp
````
and in the GDB directory:
````
earle@amd:~/src/pico-quick-toolchain/arena.win64/binutils-gdb-gnu/gdb$ grep enable-static config.log 
  $ /workdir/repo/binutils-gdb-gnu/gdb/configure --srcdir=/workdir/repo/binutils-gdb-gnu/gdb --cache-file=./config.cache --with-gnu-as --with-gnu-ld --prefix=/workdir/arm-none-eabi.win64 --disable-shared --with-newlib --enable-threads=no --disable-__cxa_atexit --disable-libgomp --disable-libmudflap --disable-nls --without-python --disable-bootstrap --enable-languages=c,c++ --disable-lto --enable-static=yes --disable-libstdcxx-verbose --disable-decimal-float --with-cpu=cortex-m0plus --with-no-thumb-interwork --with-libgmp-prefix=/workdir/arena.win64/gmp --program-transform-name=s&^&arm-none-eabi-& --disable-option-checking --disable-year2038 --build=x86_64-linux-gnu --host=x86_64-w64-mingw32 --target=arm-none-eabi
Configured with: ../../src/configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --prefix=/usr --enable-shared --enable-static --disable-multilib --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --libdir=/usr/lib --enable-libstdcxx-time=yes --with-tune=generic --with-headers=/usr/x86_64-w64-mingw32/include --enable-version-specific-runtime-libs --enable-fully-dynamic-string --enable-libgomp --enable-languages=c,c++,fortran,objc,obj-c++,ada --enable-lto --with-plugin-ld --enable-threads=win32 --program-suffix=-win32 --program-prefix=x86_64-w64-mingw32- --target=x86_64-w64-mingw32 --with-as=/usr/bin/x86_64-w64-mingw32-as --with-ld=/usr/bin/x86_64-w64-mingw32-ld --enable-libatomic --enable-libstdcxx-filesystem-ts=yes
Configured with: ../../src/configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --prefix=/usr --enable-shared --enable-static --disable-multilib --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --libdir=/usr/lib --enable-libstdcxx-time=yes --with-tune=generic --with-headers=/usr/x86_64-w64-mingw32/include --enable-version-specific-runtime-libs --enable-fully-dynamic-string --enable-libgomp --enable-languages=c,c++,fortran,objc,obj-c++,ada --enable-lto --with-plugin-ld --enable-threads=win32 --program-suffix=-win32 --program-prefix=x86_64-w64-mingw32- --target=x86_64-w64-mingw32 --with-as=/usr/bin/x86_64-w64-mingw32-as --with-ld=/usr/bin/x86_64-w64-mingw32-ld --enable-libatomic --enable-libstdcxx-filesystem-ts=yes
configure:33982: running /bin/bash /workdir/repo/binutils-gdb-gnu/gdb/testsuite/configure --disable-option-checking '--prefix=/workdir/arm-none-eabi.win64'  '--with-gnu-as' '--with-gnu-ld' '--disable-shared' '--with-newlib' '--enable-threads=no' '--disable-__cxa_atexit' '--disable-libgomp' '--disable-libmudflap' '--disable-nls' '--without-python' '--disable-bootstrap' '--enable-languages=c,c++' '--disable-lto' '--enable-static=yes' '--disable-libstdcxx-verbose' '--disable-decimal-float' '--with-cpu=cortex-m0plus' '--with-no-thumb-interwork' '--with-libgmp-prefix=/workdir/arena.win64/gmp' '--program-transform-name=s&^&arm-none-eabi-&' '--disable-year2038' '--build=x86_64-linux-gnu' '--host=x86_64-w64-mingw32' '--target=arm-none-eabi' 'build_alias=x86_64-linux-gnu' 'host_alias=x86_64-w64-mingw32' 'target_alias=arm-none-eabi' 'CC=x86_64-w64-mingw32-gcc' 'CFLAGS=-I/workdir/arm-none-eabi.win64/include -I/workdir/arena.win64/cross/include -pipe -g -O2    ' 'LDFLAGS= -L/workdir/arm-none-eabi.win64/lib -L/workdir/arena.win64/cross/lib -Wl,--stack,12582912' 'CXX=x86_64-w64-mingw32-g++' 'CXXFLAGS=-pipe -g -O2    ' 'MAKEINFO=makeinfo --split-size=5000000' 'PKG_CONFIG_PATH=' 'YACC=bison -y' --cache-file=.././config.cache --srcdir=/workdir/repo/binutils-gdb-gnu/gdb/testsuite
````

Let me do some digging.  I seem to remember having other issues when trying to ship MINGW DLLs (for openocd)...

earlephilhower, 2023-09-17:
Adding `--disable-interprocess-agent` as suggested in some of the GNU mailing lists for the Windows builds also doesn't seem to change anything WRT dependencies.
````
I have no name!@64a446016447:/workdir/arena.win64/binutils-gdb-gnu/gdb$ grep interprocess config.log
  $ /workdir/repo/binutils-gdb-gnu/gdb/configure --srcdir=/workdir/repo/binutils-gdb-gnu/gdb --cache-file=./config.cache --with-gnu-as --with-gnu-ld --prefix=/workdir/arm-none-eabi.win64 --disable-shared --with-newlib --enable-threads=no --disable-__cxa_atexit --disable-libgomp --disable-libmudflap --disable-nls --without-python --disable-bootstrap --enable-languages=c,c++ --disable-lto --disable-interprocess-agent --enable-static=yes --disable-libstdcxx-verbose --disable-decimal-float --with-cpu=cortex-m0plus --with-no-thumb-interwork --with-libgmp-prefix=/workdir/arena.win64/gmp --program-transform-name=s&^&arm-none-eabi-& --disable-option-checking --disable-year2038 --build=x86_64-linux-gnu --host=x86_64-w64-mingw32 --target=arm-none-eabi
configure:33982: running /bin/bash /workdir/repo/binutils-gdb-gnu/gdb/testsuite/configure --disable-option-checking '--prefix=/workdir/arm-none-eabi.win64'  '--with-gnu-as' '--with-gnu-ld' '--disable-shared' '--with-newlib' '--enable-threads=no' '--disable-__cxa_atexit' '--disable-libgomp' '--disable-libmudflap' '--disable-nls' '--without-python' '--disable-bootstrap' '--enable-languages=c,c++' '--disable-lto' '--disable-interprocess-agent' '--enable-static=yes' '--disable-libstdcxx-verbose' '--disable-decimal-float' '--with-cpu=cortex-m0plus' '--with-no-thumb-interwork' '--with-libgmp-prefix=/workdir/arena.win64/gmp' '--program-transform-name=s&^&arm-none-eabi-&' '--disable-year2038' '--build=x86_64-linux-gnu' '--host=x86_64-w64-mingw32' '--target=arm-none-eabi' 'build_alias=x86_64-linux-gnu' 'host_alias=x86_64-w64-mingw32' 'target_alias=arm-none-eabi' 'CC=x86_64-w64-mingw32-gcc' 'CFLAGS=-I/workdir/arm-none-eabi.win64/include -I/workdir/arena.win64/cross/include -pipe -g -O2    ' 'LDFLAGS= -L/workdir/arm-none-eabi.win64/lib -L/workdir/arena.win64/cross/lib -Wl,--stack,12582912' 'CXX=x86_64-w64-mingw32-g++' 'CXXFLAGS=-pipe -g -O2    ' 'MAKEINFO=makeinfo --split-size=5000000' 'PKG_CONFIG_PATH=' 'YACC=bison -y' --cache-file=.././config.cache --srcdir=/workdir/repo/binutils-gdb-gnu/gdb/testsuite

I have no name!@64a446016447:/workdir/arena.win64/binutils-gdb-gnu/gdb$ objdump -p gdb.exe   | grep "DLL Name:"
	DLL Name: ADVAPI32.dll
	DLL Name: KERNEL32.dll
	DLL Name: msvcrt.dll
	DLL Name: USER32.dll
	DLL Name: WS2_32.dll
	DLL Name: libgcc_s_seh-1.dll
	DLL Name: libstdc++-6.dll
````

maxgerhardt, 2023-09-17:
Hm, I'll see if I can't pass `-static` into the LINKFLAGS directly to make it statically built, same for `-static-libgcc`.. Or maybe it has to be just `--enable-static`? 

For what it's worth, running `Depencies.exe -imports arm-none-eabi-gdb.exe`  on this shows the functions, and it's just some `std::` containers and exception handling that is imported..

```
Import from module libgcc_s_seh-1.dll :
         Function _Unwind_Resume
         Function __emutls_get_address
         Function __popcountdi2
Import from module libstdc++-6.dll :
         Function _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findEPKcyy
         Function _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findEcy
         Function _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE5rfindEPKcyy
         Function _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE5rfindEcy
         Function _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEPKc
         Function _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEyyRKS4_
         Function _ZNKSt8__detail20_Prime_rehash_policy11_M_next_bktEy
         Function _ZNKSt8__detail20_Prime_rehash_policy14_M_need_rehashEyyy
         Function _ZNKSt9bad_alloc4whatEv
         Function _ZNKSt9type_infoeqERKS_
         Function _ZNSt6chrono3_V212steady_clock3nowEv
         Function _ZNSt6chrono3_V212system_clock3nowEv
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_replaceEyyPKcy
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructEyc
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE14_M_replace_auxEyyyc
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4swapERS4_
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendEPKc
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6assignEPKc
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEyc
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7replaceEyyPKcy
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7reserveEy
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE8_M_eraseEyy
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_appendEPKcy
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERyy
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_mutateEyyPKcy
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9push_backEc
         Function _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEaSEOS4_
         Function _ZNSt8__detail15_List_node_base7_M_hookEPS0_
         Function _ZNSt8__detail15_List_node_base9_M_unhookEv
         Function _ZNSt9bad_allocD2Ev
         Function _ZSt11_Hash_bytesPKvyy
         Function _ZSt17__throw_bad_allocv
         Function _ZSt18_Rb_tree_decrementPSt18_Rb_tree_node_base
         Function _ZSt18_Rb_tree_incrementPKSt18_Rb_tree_node_base
         Function _ZSt18_Rb_tree_incrementPSt18_Rb_tree_node_base
         Function _ZSt19__throw_logic_errorPKc
         Function _ZSt19uncaught_exceptionsv
         Function _ZSt20__throw_length_errorPKc
         Function _ZSt24__throw_out_of_range_fmtPKcz
         Function _ZSt25__throw_bad_function_callv
         Function _ZSt28_Rb_tree_rebalance_for_erasePSt18_Rb_tree_node_baseRS_
         Function _ZSt29_Rb_tree_insert_and_rebalancebPSt18_Rb_tree_node_baseS0_RS_
         Function _ZSt7nothrow
         Function _ZSt9terminatev
         Function _ZTVN10__cxxabiv117__class_type_infoE
         Function _ZTVN10__cxxabiv119__pointer_type_infoE
         Function _ZTVN10__cxxabiv120__function_type_infoE
         Function _ZTVN10__cxxabiv120__si_class_type_infoE
         Function _ZTVN10__cxxabiv121__vmi_class_type_infoE
         Function __cxa_allocate_exception
         Function __cxa_begin_catch
         Function __cxa_end_catch
         Function __cxa_free_exception
         Function __cxa_guard_abort
         Function __cxa_guard_acquire
         Function __cxa_guard_release
         Function __cxa_pure_virtual
         Function __cxa_rethrow
         Function __cxa_throw
         Function __cxa_throw_bad_array_new_length
         Function __dynamic_cast
         Function __gxx_personality_seh0
```

earlephilhower, 2023-09-17:
I have build GDB and manually copied the appropriate DLLs from the docker image.  Would someone be able to run it on a failing Windows PC (i.e. just from the command line, not in P.IO since it's just the binary itself)?

[gdb-test.zip](https://github.com/earlephilhower/pico-quick-toolchain/files/12643904/gdb-test.zip)

maxgerhardt, 2023-09-18:
The `gdb.exe` is startable under Windows 10, not yet checked Windows 11.  When I rename any of the three `.dlls` to a different name, they all respectively cause a "<original dll> was not found" error, which is good, so this is the minimal set and all delivered DLLs are absolutely required.

earlephilhower, 2023-09-18:
Great, thanks.  I'm traveling for business this week, but will do the same for win32 and then update the release manually and change the makefile this weekend.

maxgerhardt, 2023-09-18:
>not yet checked Windows 11

FYI, in the clean Windows 11 VM I get the same behavior, `gdb.exe` starts up normally with the delivered DLLs, renaming any one of the DLLs in the folder to something else breaks it, as is expected.

earlephilhower, 2023-09-23:
I've tried the same in a clean VM on the 32 and 64 bit versions, so this should be all resolved in the next release.

https://github.com/earlephilhower/pico-quick-toolchain/commit/fb4d23a81788bfbec0639f003ad30996c8266f23 will now automatically pull in the needed DLLs on future builds.

closed 2023-09-23 by earlephilhower