earlephilhower/pico-quick-toolchain #31: GDB crashes when using JLink

maxgerhardt, 2023-10-07:
Per https://github.com/maxgerhardt/platform-raspberrypi/issues/39, with the latest toolchain version, there seems to be a problem with the interaction of GDB and a GDB server opened by the native JLink tools (`JLinkGDBServerCLExe`) to the point that GDB just crashes.

```
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = jlink
PlatformIO: Initializing remote target...
Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x10003A40 (Data = 0x4A391AD3)
Reading register (MSP = 0x20041F00)
Reading register (PSP = 0xFFFFFFFC)
Reading 64 bytes @ address 0x20041F40

Fatal signal: 
Abort trap: 6

----- Backtrace -----
Backtrace unavailable
---------------------
A fatal error internal to GDB has been detected, further
debugging is not possible.  GDB will now terminate.

This is a bug, please report it.
  For instructions, see:
<https://www.gnu.org/software/gdb/bugs/>
.

GDB closed TCP/IP connection (Socket 11)
```

Using the *the same J-Link tools, commandline options and GDB commands* but with the original toolchain-gccarmnoneeabi (aka GCC/GDB 9.3.1) does work. So the fault seems to lie in GDB here.

Unfortunately, no backtrace is available. Is it possible to trigger a toolchain build with the debug and backtrace options enabled so that we can narrow this down further?

earlephilhower, 2023-10-07:
Couple questions:
* Is this a Mac only thing (I *think* it is from the other thread, but not sure)?
* What version of JLink is being used?  Maybe it's just too old to work with the latest GDBs?

I can try adding `--enable-backtrace` to the GDB build and not `strip` the output, but I don't think I can do that for anything other than x86-64 or i686 archs as there are external libs that probably don't exist in the cross-compilation environment for Macs.

earlephilhower, 2023-10-07:
Here's the x86_64 Linux version built with `--enable-backtrace` and un-`strip`ped:
https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.1.0-a/gdb-backtrace.zip

earlephilhower, 2023-10-07:
FWIW, here's the build for OSX:
https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.1.0-a/gdb-osx-backtrace.tar.gz

(note these only have a GDB executable so use the original tarballs to get the toolchain and then manually copy over the updated `gdb` to `bin/arm-none-eabi-gdb` in the installed dir)

GeminiServer, 2023-10-07:
1. I used the original tarball! ``` platform_packages =  toolchain-rp2040-earlephilhower@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.1.0-a/x86_64-apple-darwin15.arm-none-eabi-d3d2e6b.230911.tar.gz```
--> File: ```11M 12 Sep 02:12 arm-none-eabi-gdb```

3. I used the ```gdb``` from gdb-osx-backtrace.tar.gz and replaced the ```bin/arm-none-eabi-gdb```file. 
--> File: ```  15M  7 Okt 17:11 arm-none-eabi-gdb```
4. Then started the debug process again. Sorry, but i don't see any difference. 

<details> <summary>Console - Debug with 2.1.0</summary>

```console
Read register 'r0' (4 bytes) from hardware: 0x00000000
Read register 'r1' (4 bytes) from hardware: 0x00000000
Read register 'r2' (4 bytes) from hardware: 0x440100D0
Read register 'r3' (4 bytes) from hardware: 0x00000000
Read register 'r4' (4 bytes) from hardware: 0x0D000000
Read register 'r5' (4 bytes) from hardware: 0x081D0020
Read register 'r6' (4 bytes) from hardware: 0x00000018
Read register 'r7' (4 bytes) from hardware: 0xE81F0420
Read register 'r8' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r9' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r10' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r11' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r12' (4 bytes) from hardware: 0xC9020020
Read register 'sp' (4 bytes) from hardware: 0x980D0420
Read register 'lr' (4 bytes) from hardware: 0x0D3C0010
Read register 'pc' (4 bytes) from hardware: 0x083C0010
Read register 'xpsr' (4 bytes) from hardware: 0x00000041
Read 4 bytes @ address 0x10003C08 (Data = 0xFA5AF007)
Reading register 'msp' = 0x20040D98
Reading register 'psp' = 0xFFFFFFFC
Reading 64 bytes @ address 0x20040D80

Fatal signal: 
Abort trap: 6

----- Backtrace -----
Backtrace unavailable
---------------------
A fatal error internal to GDB has been detected, further
debugging is not possible.  GDB will now terminate.

This is a bug, please report it.
  For instructions, see:
<https://www.gnu.org/software/gdb/bugs/>
.

GDB closed TCP/IP connection (Socket 12)
undefinedLoop <_UnixSelectorEventLoop running=False closed=True debug=False> that handles pid 52268 is closed
```
</details>

<details> <summary>Console - Debug with new compiled gdb</summary>

```console
Read register 'r0' (4 bytes) from hardware: 0x00000000
Read register 'r1' (4 bytes) from hardware: 0x00000000
Read register 'r2' (4 bytes) from hardware: 0x440100D0
Read register 'r3' (4 bytes) from hardware: 0x00000000
Read register 'r4' (4 bytes) from hardware: 0x0D000000
Read register 'r5' (4 bytes) from hardware: 0x081D0020
Read register 'r6' (4 bytes) from hardware: 0x00000018
Read register 'r7' (4 bytes) from hardware: 0xE81F0420
Read register 'r8' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r9' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r10' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r11' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r12' (4 bytes) from hardware: 0xC9020020
Read register 'sp' (4 bytes) from hardware: 0x980D0420
Read register 'lr' (4 bytes) from hardware: 0x0D3C0010
Read register 'pc' (4 bytes) from hardware: 0x083C0010
Read register 'xpsr' (4 bytes) from hardware: 0x00000041
Read 4 bytes @ address 0x10003C08 (Data = 0xFA5AF007)
Reading register 'msp' = 0x20040D98
Reading register 'psp' = 0xFFFFFFFC
Reading 64 bytes @ address 0x20040D80

Fatal signal: 
Abort trap: 6

----- Backtrace -----
Backtrace unavailable
---------------------
A fatal error internal to GDB has been detected, further
debugging is not possible.  GDB will now terminate.

This is a bug, please report it.
  For instructions, see:
<https://www.gnu.org/software/gdb/bugs/>
.

GDB closed TCP/IP connection (Socket 12)
undefinedLoop <_UnixSelectorEventLoop running=False closed=True debug=False> that handles pid 48938 is closed
```
</details>

And i did check that the new file is used. Just removed to old one and start the debug!
```
Listening on TCP/IP port 2331
Connecting to target...
undefinedFileNotFoundError: 
....
```

earlephilhower, 2023-10-07:
Thanks, as I figured the Mac cross-compile environment doesn't support backtrace.  Not really sure anything can be done here in that case.  @maxgerhardt any other ideas?

GeminiServer, 2023-10-07:
I already tried to build the [pico-quick-toolchain](https://github.com/earlephilhower/pico-quick-toolchain) on macOS, but unfortunately i got an error.

earlephilhower, 2023-10-07:
Yes, this `Makefile` assumes a Linux X64 host and both the BSD nature and the different tool options (even simple CLIs commands don't match their Linux brethren on BSD).

I don't suppose there is anywhere else to obtain a prebuilt GDB 13 copy from that supports ARM?  Maybe from `brew`?

I'm still not sure what the end goal here is, though.  GDB's obviously crashing and there are no special patches or anything being applied in this build.  JLink may not be compatible w/this V13, but that should result in something other than a GDB panic.  Even with a GDB backtrace the best I think could be done would be to post something to the GNU mailing lists and their Bugzilla and hope for a debug/fix in a future GDB revision.

GeminiServer, 2023-10-07:
There is gdb release for macOS. 
```console
gdb --version
GNU gdb (GDB) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```
Quick & Dirty way: Just linked to it
```
arm-none-eabi-gdb -> /usr/local/bin/gdb
```
And i can use the debugger! It give an error in the debugger, but it works. 

```Python Exception <class 'NameError'>: Installation error: gdb._execute_unwinders function is missing
Read 4 bytes @ address 0x1000324C (Data = 0x39013820)```

<details> <summary>Console - Debug with linked to local gdb</summary>

```console
Connecting to target...
Halting core...
undefinedPython Exception <class 'ModuleNotFoundError'>: No module named 'gdb'
/Users/ecolak/.platformio/packages/toolchain-rp2040-earlephilhower/bin/arm-none-eabi-gdb: warning:
Could not load the Python gdb module from `/Users/ecolak/.platformio/packages/toolchain-rp2040-earlephilhower/share/gdb/python'.
Limited Python support is available from the _gdb module.
Suggest passing --data-directory=/path/to/gdb/data-directory.
Connected to target
Waiting for GDB connection...
Reading symbols from /Users/ecolak/Entwicklung/_Example-Projects-TEMP/arduino-wifi-scan/.pio/build/pico/firmware.elf...
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = jlink
PlatformIO: Initializing remote target...
Connected to 0000:0000:0000:0000:0000:0000:0000:0001
GDB client (conn. 12) requested target.xml from GDB Server
Reading common registers:
Read register 'r0' (4 bytes) from hardware: 0x280100D0
Read register 'r1' (4 bytes) from hardware: 0x00000000
Read register 'r2' (4 bytes) from hardware: 0x00000000
Read register 'r3' (4 bytes) from hardware: 0x280100D0
Read register 'r4' (4 bytes) from hardware: 0x388A7300
Read register 'r5' (4 bytes) from hardware: 0x00000000
Read register 'r6' (4 bytes) from hardware: 0x3E8A7300
Read register 'r7' (4 bytes) from hardware: 0x00000000
Read register 'r8' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r9' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r10' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r11' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r12' (4 bytes) from hardware: 0x00000000
Read register 'sp' (4 bytes) from hardware: 0xA81F0420
Read register 'lr' (4 bytes) from hardware: 0x03590010
Read register 'pc' (4 bytes) from hardware: 0xFA580010
Read register 'xpsr' (4 bytes) from hardware: 0x00000061
Python Exception <class 'NameError'>: Installation error: gdb._execute_unwinders function is missing
Read 4 bytes @ address 0x100058FA (Data = 0x00290020)
Reading register 'msp' = 0x20041FA8
Reading register 'psp' = 0xFFFFFFFC
Reading 64 bytes @ address 0x20041FC0
Python Exception <class 'NameError'>: Installation error: gdb._execute_unwinders function is missing
Read 4 bytes @ address 0x10005958 (Data = 0x46C0BD70)
sleep_until (t=<optimized out>) at /home/earle/src/pico-quick-toolchain/arduino/pico-sdk/src/common/pico_time/time.c:398
398	/home/earle/src/pico-quick-toolchain/arduino/pico-sdk/src/common/pico_time/time.c: No such file or directory.
Received monitor command: clrbp
Received monitor command: speed auto
Select auto target interface speed (2000 kHz)
Select auto target interface speed (2000 kHz)
Received monitor command: reset
[... 58 lines cut ...]
Read register 'r5' (4 bytes) from hardware: 0x011F0420
Read register 'r6' (4 bytes) from hardware: 0x00000018
Read register 'r7' (4 bytes) from hardware: 0x00000000
Read register 'r8' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r9' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r10' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r11' (4 bytes) from hardware: 0xFFFFFFFF
Read register 'r12' (4 bytes) from hardware: 0x1C800140
Read register 'sp' (4 bytes) from hardware: 0xE81E0420
Read register 'lr' (4 bytes) from hardware: 0x00000000
Read register 'pc' (4 bytes) from hardware: 0x82330010
Read register 'xpsr' (4 bytes) from hardware: 0x00000041
Removing breakpoint @ address 0x10003382, Size = 2
Removing breakpoint @ address 0x1000364C, Size = 2
Python Exception <class 'NameError'>: Installation error: gdb._execute_unwinders function is missing
Read 4 bytes @ address 0x10003382 (Data = 0x20202101)

Breakpoint 2, loop () at src/main.cpp:11
11	  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
Reading register 'msp' = 0x20041EE8
Reading register 'psp' = 0xFFFFFFFC
Reading 64 bytes @ address 0x20041EC0
Python Exception <class 'NameError'>: Installation error: gdb._execute_unwinders function is missing
```
</details>

maxgerhardt, 2023-10-07:
This should be much easier to reproduce without PlatformIO interfering, I should be able to just post the "to be executed commands" for arm-none-eabi-gdb and JLinkGDBServerCLExe and hopefully we get a crash + backtrace.

GeminiServer, 2023-11-11:
Any updates here?

earlephilhower, 2025-05-31:
We've moved up several GDB versions, this specific issue should be obsoleted.  4.1.0 release will have `GNU gdb (GDB) 16.0.50.20240817-git`

closed 2025-05-31 by earlephilhower