earlephilhower/pico-quick-toolchain #67: std::scoped_lock is no longer defined in <mutex>

jerfer, 2025-01-09:
IDE: Visual Studio Code (1.96.2) using PlatformIO plugin (3.3.3).

My previously building and running project no longer compiles, complaining that `scoped_lock` is not a member of `std`. This only started occurring when I upgraded Visual Studio Code and PlatformIO to their latest versions. I'm not sure whether this was the trigger for this issue or whether this is caused by pulling the latest version of the `platform` and `board_build.core`.

My research indicates that this is caused by the toolchain not defining `_GLIBCXX_HAS_GTHREADS`, which is a prerequisite for defining `std::scoped_lock` in `<mutex>`. Since this used to work, I suspect that `_GLIBCXX_HAS_GTHREADS` was previously defined by this toolchain and no longer is, for some reason.

What I've done so far:
- Upgraded Python
- Uninstalled PlatformIO
- Deleted all PlatformIO caches (.platformio, .pio)
- Reinstalled PlatformIO

earlephilhower, 2025-01-09:
Without an operating systems or threads, I'm not sure what the old calls were doing, but it was almost certainly not what you intended. 

We don't have threads, and there's no place in my build system where I've ever enabled them.

I think the issue is the move to gcc14 as required by riscv.  You probably need to look into the sdk mutexes and roll your own.  I'm not even sure if the sdk finally implementef real support for std::mutex and the like.  The RPI forums and repo would be the best place to ask them...

jerfer, 2025-01-09:
> Without an operating systems or threads, I'm not sure what the old calls were doing, but it was almost certainly not what you intended.

True. I hadn't thought of that.

> I think the issue is the move to gcc14 as required by riscv. You probably need to look into the sdk mutexes and roll your own. I'm not even sure if the sdk finally implementef real support for std::mutex and the like. The RPI forums and repo would be the best place to ask them...

Thanks for the tip, I'll dig into it. It's likely not even necessary, seeing as my code is entirely single-threaded. I'm using a library which expect these to be defined, but I'll adapt it to single-threaded execution without mutexes.

closed 2025-01-09 by jerfer