earlephilhower/pico-quick-toolchain #69: Please update release

snaens, 2025-05-23:
Please pretty please,
make a new release that includes PR #66 .

Micro-ROS on pico is borked without it,
and using old versions of [platform-raspberrypi](https://github.com/maxgerhardt/platform-raspberrypi) is driving me up a wall.

I can verify that hippo5329's patch solves the issue,
however they recently deleted their [pico-quick-toolchain](https://github.com/hippo5329/pico-quick-toolchain) fork, which had the patched release files.

Many thanks in advance,
Snaens

earlephilhower, 2025-05-23:
I've been waiting for at least a bump to the SDK, Newlib, or GCC.

You are either a) very lucky or b) very powerful because just *today* (May 23) [GCC 14.3 dropped](https://gcc.gnu.org/gcc-14/)!

earlephilhower, 2025-05-23:
Are you running Linux?  If so, try https://github.com/earlephilhower/arduino-pico/pull/2969.  I've done very basic testing with my audio, WiFi, and LLM libraries on the 2350 and 2040, but not done a full rebuild of all 18 copies of GCC yet.  That PR has the Linux X86-64 tools updated with this PR plus updated Newlib/GCC.

earlephilhower, 2025-05-24:
https://github.com/earlephilhower/arduino-pico/pull/2970

earlephilhower, 2025-05-27:
Looks like Newlib 4.4 and beyond broke the `nano-formatted-io` that we use to minimize memory and flash for `sprintf` and related calls.  CI identified the same failure mode in certain builds as another unrelated project hit:  https://sourceware.org/pipermail/newlib/2024/021118.html  That bug report didn't get any response from the Newlib team and maybe it's intentional (but breaks nano printf for C++).

earlephilhower, 2025-05-31:
@hippo5329, @snaens, Can you give https://github.com/earlephilhower/arduino-pico/pull/2975 a try?  Upgrading things took a lot of back and forth, but my stress tests for [SSL](https://github.com/earlephilhower/arduino-pico/tree/master/libraries/WiFi/examples/BearSSL_Validation), [BLE](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/MouseBLE/examples/BLECircle/BLECircle.ino), and the [PSRAM speaking LLM](https://github.com/earlephilhower/PicoLlama) now seem to work (it actually found an issue in the prior build!).  [You will need to do a `git submodule update --init --recursive` to get the new SDK directory after checking out that PR.  The Pico-SDK now references an extra module in `develop`].

I've still got a lot of other peripheral and HW tests to slog through, but it sounds like you have a good use case and need for this at hand, so your testing would be much appreciated!

hippo5329, 2025-05-31:
@earlephilhower  I will build the toolchains against latest updates and test it with micro-ROS. Thank you.

earlephilhower, 2025-05-31:
> [@earlephilhower](https://github.com/earlephilhower) I will build the toolchains against latest updates and test it with micro-ROS. Thank you.

Do you build your own toolchain or use the precompiled one from here?  If you don't have any specific need, please do try this one as it should incorporate the new define and have all the updated necessary...

hippo5329, 2025-05-31:
Yes, it is 4.1.0. My typo. :)

maxgerhardt, 2025-06-22:
Arduino-Pico 4.6.0 and the new toolchain has now been reflected in the PlatformIO platform through https://github.com/maxgerhardt/platform-raspberrypi/commit/6af38e26547c56cebdbfa4ab564eb361f7d27d4f.